yossato/virtualGimbal #2: Build error

naldoxable, 2019-08-06:
Following step by step the installation guide I'm experiencing issues in the last command `make -j4` of the secion Build

Console output attached. [out.txt](https://github.com/yossato/virtualGimbal/files/3474297/out.txt)

I've double checked all the requirement and installed everything by your guide.

I'm new to linux so please help me.

yossato, 2019-08-07:
Thank you for letting me know build error. I think i fixed the error. The commit id is 3a49be5abce20eade4b79811c5983f9a1b9e5833. I added a header include. Could you try if it is not too much trouble? There is no build error in my circumstance so that i couldn't reproduce the error.

naldoxable, 2019-08-07:
Compiled **successfully**. Thank you for fixing the issue!

closed 2019-08-07 by naldoxable

naldoxable, 2019-08-07:
I have also noticed that the calibration wont run if you do not have installed `libgtk2.0-dev` and compiled opencv with the parameter `-WITH_GTK=ON`

you should add to the instruction
`sudo apt-get install libgtk2.0-dev`
and replace
`cmake .. -DWITH_TBB=ON -DWITH_OPENGL=ON -DWITH_VTK=ON`
to
`cmake .. -DWITH_TBB=ON -DWITH_OPENGL=ON -DWITH_VTK=ON -WITH_GTK=ON` 

this worked for me and the calibration file is successfully created although i get `Gtk-Message: 17:53:31.933: Failed to load module "canberra-gtk-module"` when I start the calibration

yossato, 2019-08-09:
Thank you for an information. I fixed it. The commit id is a898d4190b2afb88bf52f9aec39fce6513df61c9 .