yossato/virtualGimbal #3: Stabilization error

naldoxable, 2019-08-07:
Sorry if I'm opening another issue but following the guide after starting the command 

`./pixelwise_stabilizer -i ~/vgdataset/syukugawara/C0003.MP4 -c ILCE-6500 -l SEL1670Z -j records/2019-04-03_07.27.36.json -z 1.3  `

I get this error message:

`pixelwise_stabilizer: /home/naldoxable/virtualGimbal/post_processing_software/src/calcShift.cpp:15: void CalcShiftFromVideo(const char*, int, Eigen::MatrixXd&, Eigen::MatrixXd&): Assertion 'cap.get(cv::CAP_PROP_FRAME_COUNT) > total_frames' failed.`

Of course I have downloaded each file and tried also to change position

naldoxable, 2019-08-09:
anyway replacing

`assert(cap.get(cv::CAP_PROP_FRAME_COUNT) > total_frames);`

to

`total_frames = cap.get(7);`

seems to work till i get another error after OpenCL initialization

`terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.1.1-dev) /home/naldoxable/opencv/modules/core/src/ocl.cpp:6460: error: (-215:Assertion failed) !alias || canCreateAlias(src) in function 'init'`

yossato, 2019-08-09:
I fixed assertion error you mentioned. It was bug. Commit id is 8d515fefe433190acc3657308e331df33567a0bf. I reproduced same error if I build a source code as Debug mode. Thank you for information.

On the other hand, I'm sorry that I didn't tell you how to install OpenCL driver for Ubuntu 16.04 that using Intel Graphics.
It is available on https://github.com/intel/compute-runtime/releases. Could you try install deb files?

I'll modify README soon.

If you have Nvidia GPU, I think you may have to install another driver like this. https://askubuntu.com/questions/846769/ubuntu-16-10-nvidia-opencl

yossato, 2019-08-09:
I added a program _angular_velocity_estimator_ that generates a JSON file. The file will be found same directory of video file after execution. 
If the file exists, the pixelwise_stabilizer starts up faster. 
To run the program, execute a command like below.
./angular_velocity_estimator ~/vgdataset/syukugawara/C0003.MP4

Multiple files are acceptable.

The angular_velocity_estimator requires long execution time so have a cup of coffee.

naldoxable, 2019-08-09:
Ok, im checking now the fix, i will report you soon.

I have a Nvidia GTX 1080 and at the beginning the program reported `OpenCL is not avaiable...` but i resolved with and easy `sudo apt install ocl-icd-libopencl1` 

Today I spend 4h in restarting _pixelwise_stabilizer_ trying to find the issue so im happy to hear that you introduced _angular_velocity_estimator_

naldoxable, 2019-08-09:
Nope, same error:
`terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.1.1-dev) /home/naldoxable/opencv/modules/core/src/ocl.cpp:6460: error: (-215:Assertion failed) !alias || canCreateAlias(src) in function 'init'`

Tried  with and without the _angular_velocity_estimator_

naldoxable, 2019-08-09:
My laptop has an nvidia graphics card and an intel integrated. When the stabilizer starts it shows only the intel integrated and everything works. But on my desktop which has only an nvidia it reports the error above. The issue seems to be fixed from your side, its only driver related. I will update you if i will find the fix. Now i can start [my project](https://github.com/naldoxable/GoStaby-Pro).
Thanks for everything

closed 2019-08-09 by naldoxable