yossato/virtualGimbal #4: Segmentation fault

naldoxable, 2019-08-09:
Using this [video](https://drive.google.com/open?id=1RBgHimN7ysZmfxNInXRTqhwom2snRcSV) i get this error:

```
naldoxable@naldoxable-linux:~/virtualGimbal/build$ ./camera_calibration -i calibration.MP4 -c gopro -l 2_7K_superview -r 4
Number of Loaded Parameters=5
NumberOfCaptureImage	:100.000000
SizeOfQuadsX_mm	:28.500000
SizeOfQuadsY_mm	:30.600000
NumberOfInnerCornersX	:7.000000
NumberOfInnerCornersY	:10.000000
Gtk-Message: 20:53:28.097: Failed to load module "canberra-gtk-module"
Chess board detection end. Estimating parameters now. please wait...

Segmentation fault (core dump created)
```
Can you try using camera name _gopro_ and lens name _2_7k_superview_ ?

Orientation is something to implement i think, see [image]

but shouldn't be the issue point.

naldoxable, 2019-08-09:
Also note that in order to install **ffmpeg** `libmp3lame` is needed.
Please add in the readme

`sudo apt install libmp3lame-dev`

naldoxable, 2019-08-13:
Found the problem, if I export the video without the metadata (eg. removing them using premiere pro/media encoder) the calibration is done without problems. its just a bug parsing the .mp4 file

yossato, 2019-08-27:
Thank you for an information. I reproduced same segmentation fault during calibration of your video in my environment. I think this is the bug of .mp4 parsing as you said. I cannot fix this bug since this bug is inside OpenCV or FFMpeg backend. I added error handling message that lets you know what happened.

closed 2019-08-27 by yossato