yossato/virtualGimbal #5: Again segmentation fault

naldoxable, 2019-08-30:
Hi, i'm still failing to stabilize the video from the GoPro. Can you help me finding the issue?

In the folder linked below, you can find:
* gyro.json (gyro data/angular velocity formatted as your software requires)
* calibration.mp4 (video without metadata to be used to calibrate internal parameters)
* test.mp4 (raw video from gopro in case you need)
* test_no_metadata.mp4 (rendered video to remove metadata and avoid parsing bug)
* cameras.json (calibration made by me using rotation 4 but probably is wrong, use gopro as name of camera and lens)

If you use the video without metadata you will be able to generate the estimated angular velocity but it fails the stabilization.

[Folder on Google Drive](https://drive.google.com/open?id=1o23Yu1qwoCitT0hj2l8CiHV5DFkzjrMg)

I will appreciate so much if you can help me

yossato, 2019-09-02:
Let me check your data.

yossato, 2019-09-10:
I'm sorry to have kept you wait for long time. I found a issue that if input video is shorter than about 64 sec, pixelwise_stabilizer failing to stabilize the video in case of the video fps is 30 Hz. I'm fixing the issue now. If it is not much trouble, could you take a longer video again and try stabilizing?

After I fixed the issue, Ill report it here.

naldoxable, 2019-09-10:
Hi, no worries! You're making a really cool project.

I've tested it with a 72s video and it worked but of course no stabilization is done and some artifacts are introduced. This is for sure due to the wrong gyro rotation. Can you explain me how to add another rotation? I've checked the source code but i don't know anything about Quaternion :(