yossato/virtualGimbal #7: Arduino version

hyperion11, 2020-07-08:
Hello! Can i ask some questions to reproduce your SD adapter in arduino?
1. As i understand SD FW simply logging angular radian per second data from gyro to json file with 240Hz frequency?
2. Record gyro data starts when video file recording starts? Or what mechanism of synchronization gyro data with video data?
I want try to make arduino version to use with RC models such as plane\copter with split cameras such as caddx turtle, runcam hybrid and others...

yossato, 2020-07-13:
1.Yes

2.Recording gyro data starts when a camera is turned on. I mean the VirtualGimbal records angular velocity evely time when there is power supply from SD card bus. When power supply is available from USB, the VirtualGimbal runs as a data transfer mode. There are two power supply mode. 
Synchronization strategy is a brute force. A post-process software calculates a correlation coefficients using sum of absolute difference (SAD) between an estimated angular velocity from video and a measured angular velocity using the gyro sensor. The measured angular velocity data is longer than estimated angular velocity data so that the post processing software calculate the SAD while changing offset time of both angular velocities. There correlation coefficient becomes the smallest value when the offset time indicates synchronized time difference between the gyro data and the video data.

wenzhicode, 2020-10-08:
Hi, yossato.

Can you tell me a little more about SAD, or is Estimated Angular Velocity from Video only used for synchronization?
Does SyncTable store multiple synchronization points?

Thanks.

yossato, 2020-10-10:
Hi wenzhicode,

The estimated angular velocity is not accurate value since it contains a translation motion of camera, a moving object in the video and mismatch of key points. On the other hand the gyro sensor’s angular velocity is more precise than estimated angular velocity. It is enough good to stabilize a video.

wenzhicode, 2020-10-10:
Thank you ,now I understand the difference.

yossato, 2020-10-10:
SyncTable stores the synchronization points. The camera clock source and virtualgimbal clock source are different so that these clock runs independently. If we synchronize these video at a beginning, the clock frequency drifts slightly so that video stabilization fails after few minutes.

yossato, 2020-10-10:
Periodic synchronization ensures the best stabilization quality. It compensates a clock frequency drifts. A default period is 30 seconds.

wenzhicode, 2020-10-11:
Thank you for your detailed explanation. I should  go back to the code and verify it with your example.

closed 2020-10-16 by yossato

hyperion11, 2020-11-30:
hi, can you share example of SD Card log?
i can't understand how it looks :(
thank you!

naorunaoru, 2020-11-30:
@hyperion11 

```
{
  frequency: number,
  angular_velocity_rad_per_sec: [
    [
      x,
      y,
      z,
      x,
      y,
      z,
      ...
      z
    ],
    [
      ...
    ]
  ],
}
```

`angular_velocity_rad_per_sec` is a nested array where innermost array may be any length.