yossato/virtualGimbal #9: About Rotation matrix

wenzhicode, 2020-09-29:
Hi, yossato.

I see here that the matrix is computed for each row.
So the transformation matrix transform each row each frame , isn't that a lot of computation? 

https://github.com/yossato/virtualGimbal/blob/master/post_processing_software/src/multi_thread_video_writer.cpp#L230

yossato, 2020-09-29:
Hi wenzhicode,
Precise rolling shutter wobble correction requires different rotation matrix in each row. 
It requires much more computational cost but offers a great stabilization ability. 

On the other hand, it is common to divide the screen into multiple areas and calculate the rotation matrix at each vertex. The method uses OpenGL and Vertex shader.

wenzhicode, 2020-09-29:
Thank your yossato.

How can I debug it? I mean, can I debug it to get a better understanding of the code.
I've been looking at the code for a long time today, and the formulas are a headache.
Sorry, I don't know much about it.
Maybe I could take a video of you and turn it into a frame of a image, and compare the difference between before and after the conversion.

wenzhicode, 2020-09-29:
Hi, yossato

https://github.com/yossato/virtualGimbal/blob/master/post_processing_software/src/rotation_param.cpp#L253

What do we mean by the first/second variable?
It's a vector on two different frames, used to calculate the Angle between the two frames, and then generate a quaternion based on that Angle, right?

closed 2020-10-10 by wenzhicode

yossato, 2020-10-10:
I’m sorry for late reply. This code calculates subframe rotation angle, so this line is interpolation between deferent frame angles.