yossato/virtualGimbal #10: Is it possible to port/build this under windows?

attilafustos, 2020-10-15:
I am struggling to build this project under windows as a console application.
Do I have a chance to to it?

yossato, 2020-10-16:
I don't have any plans for this at this time, but if it works, please send me a merge request. I'm open to questions and the request.

The basic algorithm is based on openCV and OpenCL, and is independent of the OS. However, the input/output relationship of the file may depend on the OS. Also, the backend of OpenCV video may be something other than ffmpeg.
In any case, you will write a separate cases in CMakeLists.txt for different OS.

attilafustos, 2020-10-17:
I built it on windows with minimal changes in the code.
I am currently trying to stabilize the test videos given.
I noticed poor performance on my Ryzen 5 laptop.
0.3 - 4 FPS while the GPU and CPU is low. GPU memory at 90%.
Can I improve this by changing the queue_size ?
Thank you!
[image]

yossato, 2020-10-18:
Thank you for your outstanding work.  
I think the virtualgimbal doesn't require 1GB GPU Memory. It might be memory leak that doesn't be reproduced in ubuntu enviroment. OpenCL on OpenCV is unstable. There might be available work around.  
The queue_size is a parameter for video writing. It affects video saving performance slightly but if you use fast storage like SSD, it doesn't matter.

yossato, 2020-10-28:
@attilafustos  I’m interested in running VirtualGimbal on Windows now. I’d be happy to see your modification and verify operation if you share your modified code.

attilafustos, 2020-10-28:
@yossato 
I will contribute with my modifications. I added the dependencies directly to Visual Studio. I will have to modify the cmake files.
At the moment I run into an open cv assertion error and I stopped researching.
I will post the error once I will get to that PC.
Thank you.

attilafustos, 2020-10-29:
This is the assertion error i got. I suspect is a version mismatch between installed / included opencv.
[image]

wenzhicode, 2020-10-31:
I used Opencv 4.5.0 on ubuntu 18.04, did not try on windows.

yossato, 2020-11-03:
@attilafustos VirtualGimbal estimate angular velocity of a video to synchronize video and gyro sensor data. The angular velocity is calculated from an optical flow of the video. The current VirtualGimbal implementation requires some feature points on the video to calculate optical flow. If your video has no feature points at specific frame, the calculation of optical flow will fail.

yossato, 2020-11-03:
@attilafustos I think it saves your time to run VirtualGimbal on windows using the sample data on this repository. It enables you to separate cause of errors whether the error comes from a porting or the video has no feature points.

yossato, 2020-11-03:
I think the error arisen at https://github.com/yossato/virtualGimbal/blob/8f2edad21bd3c0b287d3abf9dd204d667b4f246b/post_processing_software/src/calcShift.cpp#L97

attilafustos, 2020-11-03:
> @attilafustos I think it saves your time to run VirtualGimbal on windows using the sample data on this repository. It enables you to separate cause of errors whether the error comes from a porting or the video has no feature points.

yes, i already tested using the sample video and it worked as expected