yossato/virtualGimbal #11: Error: Measured angular velocity data from a gyroscope sensor is shorter than video length

attilafustos, 2020-10-21:
Hello,

I get error when I try my own generated gyro data: Error: Measured angular velocity data from a gyroscope sensor is shorter than video length
The json file is valid json. I am not sure what to check.

Thank you,
Attila

naldoxable, 2020-10-21:
Hi, there is a quick workaround for this issue. I just make sure my imu recorder starts at least 2 seconds before the video recorder and ends 2 seconds after. This trick make the right headroom for the sync process.

yossato, 2020-10-22:
Hi @attilafustos,
Could you show me angular velocity data? The frequency of angular velocity seems incorrect. Length of angular velocity is calculated using number of frames and frequency.

attilafustos, 2020-10-22:
Thank you for your reply.
The problem was that I had an excess comma after the last gyro value. I will test later on my desktop pc because my Ryzen 5 laptop is incompatible.

closed 2020-10-23 by attilafustos