yossato/virtualGimbal #13: Formula to calculate angular_velocity_rad_per_sec

attilafustos, 2020-10-23:
I am not sure how to calculate properly the gyro values stored in the json file.
I seen a formula in the code but I am not sure about it: 1.f/16.4f*M_PI/180.0f
The values I get from the flight controller seem to be in the -128 to 128 range.
Thank you.

yossato, 2020-10-28:
@attilafustos The formula is for MPU9250 gyro sensor.
[image]  
Your flight controller has 8bit accuracy, it is not enough accurate for video stabilization.

attilafustos, 2020-10-28:
@yossato 
Thank you for your reply.
BetaFlight compatible flight controllers use STM32 processor.

attilafustos, 2020-10-28:
@yossato 
So if I understand correctly, Gyro Raw / 16.4 gives me rotation in angles/sec.
From this I have to convert **radian_per_sec** = angle_per_sec * (PI/180) ?
1° × π/180 = 0.01745rad

wenzhicode, 2020-10-30:
yes, Yes, you are right @attilafustos

attilafustos, 2020-10-30:
@wenzhicode 
Did you encounter an error or just no stabilization on the resulting video?

wenzhicode, 2020-10-31:
I encountered an error .

Here are my steps:
1. Shooting Chesspattern videos with GoPro to get calibration parameters
2. Shoot a flight video with GoPro and separate the gyroscope data from the flight video
3. Get the predicted angular velocity data from the flight video
4. Select the placement mode of the gyroscope as 12
5. Use the data and videos obtained above to increase stability
6. Get an exception thrown

attilafustos, 2020-10-31:
@wenzhicode To get the gyro data from the flight controller I made my own app.
You need to store the data as a json file, scaled to radian/sec. Also you need to set the sampling rate on json to sampling rate used to get gyro data. I set it to 1000 hz. The exception I got from the stabilizer is assertion error. I have another issue open where I posted a screenshot.

fgiudice98, 2020-10-31:
Hi all. I just stumbled across this repo doing research for making a tool just like this for my quadcopter using gyro data from blackbox, but seems i'm not the only one that had the same idea.
BTW @yossato brilliant idea! I like the concept of this project a lot! Good job!
@attilafustos do you think you can share the tool to convert from blackbox to json? Maybe a github repo so we can help eachother?

wenzhicode, 2020-10-31:
From the [README](https://github.com/gopro/gpmf-parser), the sampling rate of the gyroscope in the gopro video is related to the model, I separated the gyroscope data directly from the video. I guess the separated data does not meet the requirements.
Yeah, I'm converting the gyroscope data to radian/ SEC.

I think you can use Python to convert the data in CSV to JSON format, and CSV files can be exported from BBL files. @fgiudice98

fgiudice98, 2020-10-31:
Yeah, i was going to make a thing like that, but since we are basically working on the same thing why not working together?

wenzhicode, 2020-10-31:
What can I do？

attilafustos, 2020-10-31:
@wenzhicode  @fgiudice98 
https://github.com/attilafustos/bf2json

fgiudice98, 2020-10-31:
@wenzhicode I think a repo with scripts to adapt and configure for the use with blackbox data along with some preconfigured cameras (such as Hero7, or Caddx integrated) can be a good start

wenzhicode, 2020-10-31:
Yes.

As I understand it, virtualGimbal needs to have the following pieces of data:
1. Calibration parameters of camera
2. Authentic gyroscope data
3. Prediction data used to find synchronization points
The correct gyroscope data can be extracted from the Blackbox using a script, and the calibration parameters of the camera should not be difficult to obtain.My main challenge is how to verify that a video meets virtualGimbal's requirements, or can we modify virtualGimbal to match our videos?

fgiudice98, 2020-11-01:
@wenzhicode what are the specs of the video you are trying to stabilize?