yossato/virtualGimbal #14: BOM File

shmsuni, 2020-11-04:
Would it be possible to share the BOM files and the Pick and Place files that you used to get it assembled? The one I have generated from Eagle seems to be missing some data and my PCB manufacturer won't accept that.
Thanks.

yossato, 2020-11-09:
@shmsuni 
I put bom lists below.
https://github.com/yossato/virtualGimbal/blob/master/circuit_adapter/bom.txt
https://github.com/yossato/virtualGimbal/blob/master/circuit_sdcard_rev.4.2/bom.txt

I'll add additional part data later.

yossato, 2020-11-09:
I made a sdcard circuit as a flexible printed circuits. A thickness of the circuit is only 0.1mm.

shmsuni, 2020-11-09:
Thank You Yossato

In the given BOM file, there is a mention of the following parts:
CN3  MICROSD_BARE
CN4  VIRTUALGIMAL 

The parts are from the project's custom library. Would it be possible to share the details of the parts (manufacturer/code) and where to source them?

Thanks

yossato, 2020-11-10:
I think connector.lbr contains both MICROSD_BARE and VIRTUALGIMBAL below. Could you open connector.lbr with EAGLE?  
https://github.com/yossato/virtualGimbal/blob/master/eagle_library/connector.lbr

[image]

[image]

yossato, 2020-11-10:
Note that A thickness of the adapter circuit is 0.6 mm.