yossato/virtualGimbal #15: Possibility to use variable gyro rate

naorunaoru, 2020-11-11:
I've cobbled together a low-cost IMU logger with AT328P and MPU6050 to use with post processing software, but it's hard to make it log 240Hz, 500Hz is ok though. That needs further investigation, because it does not use DMP/interrupts and theoretically should work at lower speed. 

But hypothetically we can approach this problem from another angle.

Is it too complicated to make the software use different gyro rate? 

Alternatively I can use pandas/scipy to resample 500Hz stream into 240Hz but I'm afraid it will cause aliasing and jitters.

Thanks.

yossato, 2020-11-11:
Hi @naorunaoru ,
500Hz is ok if you set it at frequency of angular velocity json data. See a sample data below. There is a setting at line 2. I think a higher frequency than 240Hz is more suitable since it contains a detailed camera motion.
https://www.dropbox.com/s/mz7mn3maapkgodw/2019-04-03_07.27.36.json?dl=1

[image]

naorunaoru, 2020-11-16:
Thanks @yossato, it works perfect.
Also I successfully compiled the stabilizer on a mac with some tweaks to CMakeLists.

どうもありがとう！

closed 2020-11-16 by naorunaoru