yossato/virtualGimbal #30: PCB Help

lucasheridon, 2022-06-04:
Awesome project. I would like to make one, but I am struggling to understand the instructions. First I am wondering what I need to buy and what file I am supposed to use to create the PCB.
Second, I would like to use this with GyroFlow and would like to know if the gyro data provided from this project would work with the GyroFlow program.
Thanks

yossato, 2022-06-07:
Thank you for your question.
The parts list is in the BOM at #14 .
You will need to make two different boards for the adapter and the SDcard.

I have never heard of GyroFlow before. It sounds like a great software. the gyro sensor data acquired by VirtualGimbal does not have a time stamp. So we need to synchronize the video and the gyro sensor, and if GyroFlow has a synchronization function, maybe it could be used.
VirtualGimbal's post-processing software has a built-in synchronization function.

CptCurk, 2022-09-02:
Hey, I come from Gyroflow's software page also. And your SD card looks great ! 
I'm a professionnal filmmaker and would love to use your hardware for recording the gyrodata (almost sync isn't it ?) and then work it with Gyroflow (in macOS environnement, I don't have time to learn and use Ubuntu, I did try ^^'). I don't really understand how to work with BOM. How I can design a working PCB and Gerber file ? Where could I print that ?

But before everything, your datafile is a *.CSV file ? how it's working on the SD card ? Each time it receive data from the camera, it begin to record the gyrodata and when camera data stop, it stop also ? 

I'm trying to understand everything by myself, but I don't succed there... 

Thank you for helping ! 👍

yossato, 2022-09-09:
The data file is json that contains angular velocity of a camera. It begins recording the camera rotation when camera turns on. It ends when the camera turns off. 

I’m now trying make a iPhone app that records gyro angular velocity. It is easy to get because you might have iPhone and there is App Store . 
You will attach your iPhone to your camera using 3D printed adapter, run the app to record camera motion. 
Is there a need from professional filmmakers like you?

CptCurk, 2022-09-29:
Hey, I'm actually using an iphone 12 mini for Gyroflow tried with 3 apps, sensor logger has been the best. The only problems occuring there, it's locked on 100Hz. And that's not enough, even for slow shot handheld.

I find that guy and that project : https://github.com/alexeystn/tiny-blackbox
It's kinda like your idea, but he's not making it inside an SD card. But it looks like it could fit. So both idea, and everyhting would be a dream for many videogrpaher ! ^^

Actually, I know that some movies productions used SteadXP, I even helped one to use the soft. So, yes, for some productions using gimbal and on top of it using steadxp for rolling shutter anihilation (better 3D camera tracking) and having the ability to lock horizon is something big productions are looking for.
or filmmaker doing run&gun basic interview content in the street, or even think about those war videographer, travel adventures filmmaker (reaching for light gear), having the ability to stabilize in post properlly some shots, it's a big comfort and a plus ! 
Of course, we don't "need" it. But if the option's there, it's greeeat !

yossato, 2022-12-29:
@CptCurk 
Thank you for your impressive comment.
I have made the iPhone app and recorded a camera motion.    
    
[image]    
As you said the sampling rate is locked on 100Hz. I tested video stabilization using angular velocity from iPhone. I found that precise and accurate time synchronization between angular velocity and video is essentially important for better stabilization. I improved the synchronization algorithm and got result below. Is a quality enough good for video grapher?

https://youtu.be/vz0GU34rXmc

CptCurk, 2022-12-30:
@yossato hey ! That’s pretty good actually, may be walking a little bit softer and allowing more crop.
Also looking at increasing shutter speed value (trying with 1/1000 to be sure)

then it would be perfect 👍 

i would love to learn and test your soft actually, but I’m not good enough in informatic for that unfortunately 😅