yossato/virtualGimbal #31: Support for Gyroflow

WriteCodeEveryday, 2022-09-06:
@yossato, are you still doing this project?

I'm interested in this with the rise of Gyroflow and would love to buy one unit or figure out how to put one together for Gyroflow.

yossato, 2022-09-09:
I’m doing this project until now. I made a prototype but it consumes time to make.  There some issues that prevent this to be mass production below. 

First, a SD card housing made with nylon 3D print is easily broken. 
Second, signal quality is not good so there is a compatibility issue. Some camera doesn’t recognize it as a SD card. 
I need help to improve these problems.

WriteCodeEveryday, 2022-09-11:
@yossato 
Why not switch the housing to flexible materials like TPU? Not really shock absorbing but hard to break.
"Signal quality" like the signal quality in electronics or in the actual gyro output?
I'll help you out with anything but my electronics/PCB knowledge isn't super good.